makerbase-mks/MKS-H43 #7: H43 does not read SD card

Voxel3DPrinting, 2021-03-14:
Hey,
My H43 screen does not read the SD cards for my GCODE files, I am using a MKS SGen L V2, with MKS2209 drivers and configured marlin exactly like the Wiki said..
I also have tried multiple SD cards, formatted in different ways

Jontux, 2021-04-06:
Same here with an H43 connected to a SKR V1.4 Turbo. SD card does not appear neither with the SD card connected to an H43 nor mainboard.

reQu1em00, 2021-04-07:
I can now get it working but only with a very small SD card.  128MB worked for me.  No other size will work even though I can update firmware with a larger version.

Voxel3DPrinting, 2021-04-07:
I modified the Board file and I now can read the Gcode from the onboard SD, but the screen SD does nothing. Contacted MKS via Ali and they said the H43 cannot read Gcode..
Maybe an Idea to put this information somewheere @mks-viva @makerbase-mks ??

Jontux, 2021-04-07:
I noticed also that Marlin tryes to read the SD card from pins of AUX-2 when the screen is set to MKS H43. Mainboard is SKR 1.4 Turbo.

But ok, I bought the H43 to read the SD card and to control printer but nice to know if it doesn't even read the G-code 😅 I would like to know if it's even possible to do so with some firmware changes 🤔

Voxel3DPrinting, 2021-04-08:
> I noticed also that Marlin tryes to read the SD card from pins of AUX-2 when the screen is set to MKS H43. Mainboard is SKR 1.4 Turbo.
>
> But ok, I bought the H43 to read the SD card and to control [...]

Same exact problem here.. I just grinded a slot in my printers case to get acces to the SD slot. I fixed the reading issue by changing the Board pins file and settin gthe SD mode to "ONBOAD"
Maybe this works 4 u to

Metalazzo, 2021-04-24:
I have the same problem. I cannot boot with a micro sdcard inserted in the H43 LCD (black screen, I suppose it tries to update) or if I plug it after boot, there is no file found. Read perfectly gcode files from E3D main board.
I think it's really stupid coming from MKS. And I don't understand why "it cannot read gcode". Well according to schematic it can. That they don't want to make any effort on the software side, it's an other story.
Finally, I found their lack of support disturbing... For example, you can't have a BLtouch and neopixel plugged on a MKS E3D and H43 out of the box, you need to reassign pins.

buiduy6868, 2021-06-05:
@Voxel3DPrinting 
Hi bro, have you solve your problem, I got the same issue with my H43+MKS Sgen L V2. I dont know how to do now

Voxel3DPrinting, 2021-07-09:
It appears the SD on the screen is only ment for updating the H43 screen and not for data transfer.
For the Gcode files you have to use the motherboard SD

closed 2021-07-09 by Voxel3DPrinting