makerbase-mks/MKS-H43 #11: MKS H43+Robin Mano 3 stops at the logo

YuraBz, 2021-06-08:
Hi, the screen from the box not booting, I tried to refill all the firmware, configured according to the instructions, nothing helped.  After loading it stops at the mks logo.  
Connected to mks robin nano 3

Soprano80, 2021-06-13:
https://github.com/makerbase-mks/MKS-H43/wiki/Use-MKS-H43-on-different-motherboards#mks-robin-nano-v3

Marlin-Konfiguration:

Configuration.h:

#define SERIAL_PORT -1
#define BAUDRATE 115200

Configuration_adv.h:

#if HAS_DGUS_LCD
#define LCD_SERIAL_PORT 3
#define DGUS_BAUDRATE 115200

YuraBz, 2023-03-18:
[image]
sorry but its not work. i try couple version and firmware

Crackerdent, 2023-12-19:
> [image] sorry but its not work. i try couple version and firmware

Hi YuraBz, 
I have same issue.  I bought a new mks h43 display for a corexy printer and it gets stuck on the green logo. I tried to upload a few times new firmware. But it's not work. I am using with Mks Robin nano v3, same board.
How did you do it? Can you help me?

YuraBz, 2023-12-20:
Hi. Yes. I have working firmware. Try to find tomorrow and let you know

easybusy.com.ua
***@***.***
0660501282