makerbase-mks/MKS-H43 #12: Can't set english language

matthgyver, 2021-06-09:
Hi,

In the language selection menu, I show chinese and english but if I click on english item, nothing is happening. I hear bip sound but the select indicator keep on chinese item.

I'm using [your firmware](https://github.com/makerbase-mks/Marlin-V2.X-MKS-H43).

I've tried  with both of your LCD : firmware H43_Firmware_1.31 and H43_Firmware_for_Malin and the result is the same.

Could you help me please ?

Thank's

matthgyver, 2021-06-10:
It's was a serial connection problem.

closed 2021-06-10 by matthgyver

mlueft, 2021-06-20:
what‘s your solution? I habe the same problem.

regards,

michael

matthgyver, 2021-06-20:
It's was a serial connection problem. If the screen can't communicate with the board, you can't modify some parameters 
On my Rumba32 board, I've connected my screen on A9 & A10 pins and set ``#define SERIAL_PORT -1`` into my configuration.h