makerbase-mks/MKS-H43 #15: MKS H43 Temp Sensors Not Reading Properly

CollinCBoring, 2021-07-22:
I am using the Robin Nano V3 with the latest Marklin 2.0 from the Makerbase Github.  When I use the TFT35 V2. the temperatures read properly, but when I program for the H43, the temps swing between -9000 and 9000 rapidly.  The machine will still preheat and print even though the H43 is displaying the data wrong.  The Nano V3 seems to know the temps but the H43 is displaying them wrong.
[image]
[image]
[image]
[image]
[image]

CollinCBoring, 2021-07-30:
Any Progress on this issue @makerbase-mks ?

CollinCBoring, 2021-08-30:
@MKS-Sean @makerbase-mks Any Progress Here???

slapukas, 2021-09-01:
Robin Nano V3 and MKS H43 wrong  temetarure:(((