makerbase-mks/MKS-H43 #18: Unable to change language on display

josurr, 2021-08-30:
Using the H43 (v1.0) on an Ender-3 v2 with the latest Marlin 2.0.x bugfix.

I have enabled the LCD correctly (although I have only been able to get to the main menu when the LCD serial port is set to 3 and not 1 or 2).

I also noticed that the back arrow on the Print screen (with the list of the various files on the SD card) does not work.

Additionally, the +/- axis buttons are non-functional on the Move screen.

I have flashed my H43 with the non-beep version of v1.31 (from [https://github.com/makerbase-mks/MKS-H43/tree/main/DWIN_SET_For_H43/H43_Firmware_for_Malin](https://github.com/makerbase-mks/MKS-H43/tree/main/DWIN_SET_For_H43/H43_Firmware_for_Malin))

There also does not seem to be a way to switch to the blue UI, as commenting out the GREEN_UI line causes the printer to think that it is printing on boot (display shows a "Stop Print" message), and if yes/no is chosen (or no action is taken for about 5 seconds) the screen changes to the second frame of the boot "animation" and freezes.

I was hoping that this LCD screen would be an upgrade for my printer and allow me to visualize the mesh leveling graph of the print bed, but at this point, I am seriously contemplating returning the screen to Amazon.

jorgejglopez, 2023-01-20:
I too have had the same problems.  I have tried all the firmwares and the only ones that have worked well for me are those of version 1.30.

omrfrkmll, 2023-04-24:
Hello;
for "I also noticed that the back arrow on the Print screen (with the list of the various files on the SD card) does not work."
>you should edit the screen firmware with given on MKS H43 firmware ->https://www.dwin-global.com/tool/ -> T5L menu -> DGUS_V7641[DOWNLOAD](https://www.dwin-global.com/uploads/DGUS_V7641-0801.zip)
also you need download the source DWIN_SET file from -> https://github.com/makerbase-mks/MKS-H43-firmware (choose compatible one)
open DWprj.hmi
go to page that you need edit.
[image]

on left side you can see the "page switching" section
[image]
click the set button and select home page.
than upper left side under "common" menu you can see that icons
[image]

click them and save three of them. 
all done. you just copy the DWIN_SET file to your SD card and flash to the MKS H43

-I just mentioned. not explained detailly. if you need help please write me...
I hope it is helpful :)