makerbase-mks/MKS-H43 #20: H43 on Robin Nano S 1.3

Bergateo, 2021-09-26:
Hi everyone, I would like to know if is it possible to connect the display on the board mentioned on the title and how eventually.
Because I canot find the RX1 and TX1 pin on the scheme of the board nor the aux...

Cyberyoryo, 2021-12-15:
I have managed to get it to work with a SKR mini E3 V3.0 but the language cannot be changed.  After many tests in configuration_adv.h I put serial_port 1 in DGUS and serial _port 2 in configuration.h

LiamTal, 2022-06-24:
> I have managed to get it to work with a SKR mini E3 V3.0 but the language cannot be changed. After many tests in configuration_adv.h I put serial_port 1 in DGUS and serial _port 2 in configuration.h

Hi, maybe you could explain how you solved it in #24378 https://github.com/MarlinFirmware/Marlin/pull/24378