makerbase-mks/MKS-H43 #24: Can't update the H43 Firmware

neTix84, 2021-12-08:
Hello, 

i downloaded the firmware  from https://github.com/makerbase-mks/MKS-H43/tree/main/DWIN_SET_For_H43/H43_Firmware_for_Malin

I followed the instructions: 
- Copy "DWIN_SET" folder to TF card after decompression.
- Power off the MKS H43, insert the TF card into MKS H43, then power on. An automatic update interface will appear on LCD.

When i use the Firmware for Marlin: 
The update interface appears, but while loading the files ... 
CFG...1...fine
Code...0...fine
BIN...it stops

When i use the normal Version:
CFG...1...fine
Code...0...fine
BIN...4...fine
S2K...it stops

Any idea?

ngoalongcang, 2022-04-02:
some one have anser?

sherwinrastegar, 2023-09-12:
Hello,
how was your problem solved? I have this problem 
how can i upload frimware ?