makerbase-mks/MKS-H43 #26: impossible change the language

Cyberyoryo, 2021-12-22:
There is no way to make the screen work and less to change the language. I have tried with the latest marlin that brings support for the screen and it does not work with any of the three updates that you have in the repository. I have tried with the marlin that you provide, I even bought a plate that was compatible with that version of marlin, and there is no way to make it work or change the language.
The truth is that I spent € 50 on the screen and € 45 on a MKS motherboard compatible with the screen and with the marln that you provide and there is no way to make it work.
No more buying MKS products, recommending MKS and everything that has to do with MKS.

AlexStepnogorsk, 2021-12-26:
Редактируй картинки в прошивке экрана.

Cyberyoryo, 2021-12-26:
- [ ] ¿Que más quieres que edite? Le he dado 20 vueltas al firmware.

Cyberyoryo, 2021-12-31:
> 

¿Que parte del firmware tendria que editar? 
What part of the firmware would I have to edit? 
Какую часть прошивки мне придется редактировать?