makerbase-mks/MKS-H43 #30: MKS H43 v1.1 on Skr 1.4 turbo

andyrok, 2022-08-31:
I don’t understand to connect the MKS H43 with the  Skr 1.4 Turbo…
Can anyone help me?
Thanks

mlee12382, 2022-08-31:
On my skr mini I just used the 10pin cable to exp1 and adapter B for the H43. Then I assigned the LCD Serial Port in configuration_adv.h to 2 I think it was, I can check when I get home.

andyrok, 2022-09-03:
I can’t connect the H 43 to the SKR 1.4 Turbo. The suggested connection is between the AUX-1 port of the Tft card and the WiFi port of the Skr... I don't have a cable that can get into the Skr socket... I tried to connect them with single cables but the screen at most lights up with the Makerbase logo and stops...🤬

andyrok, 2022-09-09:
After a million tests nothing happens.. the screen remains on the initial logo... do you have any solution?
 the board is SKR 1.4Turbo whit TMC2029

mlee12382, 2022-09-15:
Did you define an lcd serial port in configuration_adv.h under the dgus section? Ie. #if HAS_DGUS_LCD
  #define LCD_SERIAL_PORT 6
  #define LCD_BAUDRATE 115200

andyrok, 2022-09-15:
> Did you define an lcd serial port in configuration_adv.h under the dgus section? Ie. #if HAS_DGUS_LCD
> #define LCD_SERIAL_PORT 6
> #define LCD_BAUDRATE 115200

Are you sure about 6..?
I try 👍

mlee12382, 2022-09-15:
> > Did you define an lcd serial port in configuration_adv.h under the dgus section? Ie. #if HAS_DGUS_LCD
> > #define LCD_SERIAL_PORT 6
> > #define LCD_BAUDRATE 115200
> 
> Are you sure about 6..?
> I try 👍

I don't think it matters but that's what it is on mine. I think it just needs to be different than something you're using elsewhere.

mlee12382, 2022-09-15:
[image]

andyrok, 2022-09-15:
I’ve try this option( LCD_SERIAL…) but not other number then 0…. I try tomorrow..👍