makerbase-mks/MKS-H43 #33: MKS-H43 not working MKS-Tinybee

deadstarcan, 2022-11-12:
I didnt manage to use MKS-H43 at Tinybee, what are codes/port number of that?

sherwinrastegar, 2023-09-11:
hello
, how was your problem solved? I have the same problem, and when I compile, it says that the board does not support this display.

deadstarcan, 2023-09-11:
Didnt test it yet, I will look up at weekend

sherwinrastegar, 2023-09-11:
Thank you very much, my friend
 can we communicate more?

deadstarcan, 2023-09-11:
You can email me anytime, canabanoz@gmail.com

sherwinrastegar, 2023-09-11:
thanks my friend 
my email is : shervin85rastegar@gmail.com

deadstarcan, 2023-09-11:
You are welcome :)

sherwinrastegar, 2023-09-11:
You have finished making your own printer؟

deadstarcan, 2023-09-11:
Actually I bought an anet a8 clone with fried motherboard and psu, and decided to rebuild all electric and electronic parts. Last week I have printed new hotend adapter for bmg direct drive and didnt power on yet.