makerbase-mks/MKS-H43 #35: I buy or not?

stef-ladefense, 2023-01-27:
hello makers,
I feel like MKS is completely untracking the issues here and doing nothing to fix their firmware.
exactly like their TFT24/28/32 screens etc...
Am I wrong or not?
I would like your opinion!
thank you makers

580, 2023-02-08:
I am here to see if any new firmware released, and I see your post. 
Short answer: **No**.

I have bought this for about 2 yrs, and I only have used it once or two. The reason I am here is that one of my screen(not this one) broken today, and I'd like to see if any new firmware released here so I can give it another try instead of buying a new one. And, I am going to buy a new one. But not this one.

Papadoma, 2023-04-03:
**Don't buy it**

There is no support whatsoever.

TackleBerryCrash, 2023-04-10:
DON'T

I stopped buying MKS hardware a while back and bought this before that decision. 
I have spent more time trying to figure out their weird not compatible with each other hardware from spent using it.
No proper documentation, no support, no answers.

stef-ladefense, 2023-04-10:
thank you all indeed! so it is really this brand that is not reliable.
I have a TFT24 + TFT35, with software full of bugs, I was in retreat at Bigtreetech, and did a bug hunt to be able to put their software on my tft35.
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/2732
I also stop with Makerbase